666djb/solarpi #1: Make USB Device configurable

factor4, 2023-04-24:
Hi! First of all thank you for sharing your work! I managed it yesterday to get my Growatt SPH 6000TL3 connected. 👍 

I use this USB to RS485 Adapter: https://www.amazon.de/dp/B0B87D9LNC
This adapter is attached to /dev/ttyACM0 instead of /dev/ttyUSB0. I changed all occurences in your code to my device address and it works as it should. It would be great if this could be changed via the configuration file.

One other thing I want to share. I connect my inverter at the RS485-3 Port and I have to use the pins 1 and 5 to get it connected. Found that info here: https://www.photovoltaikforum.com/thread/192228-growatt-sph-modbus-rtu-rj45-pinout-und-register-beschreibung/

I am very happy that I have now connected the inverter directly. Even if it rains today! ;) Thanks a lot!

666djb, 2023-04-28:
Hi, thanks for getting involved.

I'll add the 6000TL3 to the list of tested Growatts.

I'll also make an update so that the USB device is specified in the options.json config file and also note the alternative RS485 pin out in the readme.

Watch out for an update over the coming days.

factor4, 2023-04-28:
Hi! Great to hear! My solarpi runs for nearly a week now without any problems! 👍

666djb, 2023-04-29:
@factor4 I've published an update. Note, the config file options.json now needs an extra line in it compared with the earlier version. As always, back up your working install before testing!

factor4, 2023-05-02:
Hi! I've just updated it and it works! Thanks! :) 

But I have to copy the files manualy, because the `update.sh` script did not work:
```
pi@solarpi(rw):~/solarpi$ ./update.sh
Note updating to version 1.0.4 requires an additional line in the configuration file options.json to be added for the USB device path
Updating solarpi installation
Do you want to keep existing configuration file?
./update.sh: 6: Syntax error: "do" unexpected
```

666djb, 2023-05-07:
I had been running the update.sh script with ". ./update.sh" rather than making it executable and running it with "./update.sh", this meant that I was running it using bash and not sh, the select; do syntax only works in bash. Anyway, I've fixed the script so it uses bash by default! You can test by pulling the latest change or just editing to the first line in update.sh to read "#!/bin/bash"

factor4, 2023-05-12:
Hi! Works now! Thank you!

closed 2023-05-12 by factor4