666djb/solarpi #2: PV2 voltage shows PV2 power

factor4, 2023-05-12:
Hi! Me again. :)

There is a small typo on the following line:
https://github.com/666djb/solarpi/blob/605d63c648fbaabda62e47983dba05c3c189e832/src/growattSPH3000.ts#LL86C37-L86C37

Currently this caused it to show the power instead of the voltage of the second PV-String. IMHO this should be `value_template: "{{ value_json.vpv2 }}",`. I changed it in my installation and will confirm it tomorrow if it shows the voltage correctly. Should I create a pull request for this? Maybe a little overkill for just changing a single character. ;)

666djb, 2023-05-17:
I think you are right - good spot! I'll confirm and update over the coming days.

Thanks!

666djb, 2023-05-21:
V1.05 fixes this problem

closed 2023-10-01 by 666djb