666djb/solarpi #3: Inverter temperature not showing

Danm-89, 2023-05-15:
Hi, 
Great bit off software - love the new updates with time of use controls!
I am having a bit of an issue where the inverter temperature always reads 0c - All the other values seem correct - any help greatly appreciated. 
Inverter is SPH3000. I am using version 1.0.4. 
[image]

666djb, 2023-05-17:
Hi! Attached is what mine is showing - you can see a good range of temperatures. I'll have a look and see how we can debug this.

[image]

666djb, 2023-05-21:
There is a backup utility hidden away in my code. I've now updated this and written a readme for it: [](https://github.com/666djb/solarpi/blob/public/backup.md). If you want to run this and share the resulting text file, I'll happily have a look and see what your inverter is saying for the temperature values. Input Register #93 should have your inverter temperature in it, but it needs to be divided by 10 to covert to degrees Celsius (the value for mine is currently 499 which means 49.9C).

You'll need to update your install with my latest to use the backup utility and see the backup.md readme file.

Danm-89, 2023-05-21:
[growattbackup.txt](https://github.com/666djb/solarpi/files/11525350/growattbackup.txt)

Backup file attached - It looks like all of the input registers are showing as 0.

666djb, 2023-05-30:
I'll see if I can put some more debugging in the backup script and maybe get it to check additional register locations, just in case.

Before that, can I check what you are running solarpi on (e.g. Raspberry Pi or other)? Also, it maybe worth a double check to ensure that solarpi is not running when you run the backup script (sudo systemctl disable solarpi) and do a reboot. Re-enable after backup with sudo systemctl enable solarpi followed by sudo systemctl start solarpi.

Also, do you get PV Voltage and Power etc. values in the ShinePhone app / on the Growatt web page? I know inverter temperature is not shown there, but Voltage etc. are other parameters accessed via the Input Registers.

Danm-89, 2023-05-30:
Thanks, I'm using an RPi 3b for SolarPi.
I made sure I had stopped the SolarPi app before running the backup script. 
I do not have Solar currently, just battery storage. I can see battery temperature as an option in ShinePhone.