666djb/solarpi #5: Port Not Open Error

Tailzer, 2023-12-01:
I have a SPH3600 inverter that ive connected to an install of solarpi on a pizero w. I get the topics coming into Home assistant but no values. I did wonder if being a 3600 was an issue but previous poster has running, (I also have configured as SPH3000).
Ive tried pins 4&5 , and 1&5 but same error. Any ideas? 
Great project and looking forward to getting it up and running.

666djb, 2023-12-02:
Hi, it should work. A few things to check:
1. Try unplugging the RS485 to USB adapter from your Pi and then plugging it in again. Run "dmesg" as root from the command line just to double check that a) the Pi has recognised the adapter and b) what USB port it has allocated. Mine is on "/dev/ttyUSB0" but other adapters might appear as "/dev/ttyACM0". I give an example in readme file taken from my Pi to show that it is has allocated "ttyUSB0" (we add the /dev/ bit on in the options.json file).
2. If you had to add the RJ45 (ethernet style) plug on the end, check that the join is good. I tried several times to crimp an RJ45 connector onto the bare ends of the USB adapter's wires and failed despite having made many ethernet cables with my crimper. I ended up taking an existing ethernet cable and chopping it off a few inches long and joining the wires (first with a screw block to test and then I soldered to make it look a little neater).
3. Which USB adapter are you using?

As you say another poster has it working with the 3600 and I understand the only difference between my Growatt SPH3000 and the SPH3600 is that you're lucky enough to have a slightly more powerful inverter!

David

Tailzer, 2023-12-02:
thank you so much for the fast reply.
I have this adapter USB-RS485 Cable, SER CONV USB-RS485-WE-1800-BT from amazon. connected to the A & B wires from the diagram on the advert.
I have tried unplugging and replugging, demsg shows the fdti adapter and confirmed its on ttyub0 as per the config. 
I also doubted my crimping so have cut a network patch cable in half and connected to that.
Is it possible I have a duff adapter? would your software say connection refused if adapter wasnt working?
The only other thing i wondered was all the posts say two wires, does it not need a ground for reference?
thanks
Gary

Tailzer, 2023-12-14:
Tried two different adapters and same issue, so had a bit more of a play today thinking my inverter may have crashed. tried to reboot but all same, checked if I get same msg without dongle installed and yes! so reinsterted and restarted service, ran journal cmd and saw that access was denied to USB, fixed that and bingo! Now getting readings into Home assistant. Cant believe I missed that error message before. Thanks for your help and of course the software.

closed 2023-12-14 by Tailzer

666djb, 2023-12-14:
Thanks for the update, and I'm glad you have it working. I'll have a look at the readme file and put something in about checking for USB device permissions.