666djb/solarpi #7: Timeout Error reading inverter data

tekgrrl, 2024-11-25:
I've configured SolarPi and wired it up to my Inverter. Details of my setup are below. I can see the USB adapter's FTDI string when I run `dmesg` and that it's created ttyUSB0 which I've using in the options.json.  The `solarpi` user is in the group that owns the device file. I've also put the RS485 ports on the Inverter into VPP mode.

The `solarpi` service says it's connected to the inverter and to MQTT on HomeAssistant. But after that it repeats the following error message:

```
Nov 25 14:28:10 rs485 npm[997]: 25/11/2024, 14:28:10 GMT Getting control values from inverter
Nov 25 14:29:01 rs485 npm[997]: 25/11/2024, 14:29:01 GMT readHoldingRegisters() modbusClient.readHoldingRegisters() error: [object Object] retrying
Nov 25 14:29:11 rs485 npm[997]: 25/11/2024, 14:29:11 GMT Error reading inverter data: Error: timeout while waiting for mutex to become available
Nov 25 14:29:11 rs485 npm[997]:     at file:///opt/solarpi/node_modules/async-mutex/index.mjs:1:19
Nov 25 14:29:11 rs485 npm[997]:     at ModuleJob.run (node:internal/modules/esm/module_job:195:25)
Nov 25 14:29:11 rs485 npm[997]:     at async ModuleLoader.import (node:internal/modules/esm/loader:336:24)
Nov 25 14:29:11 rs485 npm[997]:     at async loadESM (node:internal/process/esm_loader:34:7)
Nov 25 14:29:11 rs485 npm[997]:     at async handleMainPromise (node:internal/modules/run_main:106:12)
Nov 25 14:29:11 rs485 npm[997]: 25/11/2024, 14:29:11 GMT Publishing data
```

I've cut an RJ45 patch cable and at first wired pins 4 and 5 to the the green and white wires of the USB adapter. I've since tried it with pins 1 and 5, no difference. if I run mbpoll I get:

```
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 1
Communication.........: /dev/ttyUSB0,       19200-8E1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1... Ctrl-C to stop)
Read output (holding) register failed: Connection timed out
-- Polling slave 1... Ctrl-C to stop)
Read output (holding) register failed: Connection timed out
-- Polling slave 1... Ctrl-C to stop)
Read output (holding) register failed: Connection reset by peer
-- Polling slave 1... Ctrl-C to stop)
Read output (holding) register failed: Connection timed out
-- Polling slave 1... Ctrl-C to stop)
Read output (holding) register failed: Resource temporarily unavailable
```

Which is weird and suggests some level of randomness in the error. I've also tried with different baud rates and with a different Slave address.

**Note:** The Battery BMS uses the CAN port on the inverter. The Inverter manual says that you can't use CAN and RS485 for battery management, but I think it's fine to use the RS485 port(s) in VPP mode while the battery uses CAN,

Do you have any ideas? I'm going to keep trying different wiring configurations and maybe stick a scope on the RS485 output. But I would definitely appreciate any guidance.

**Inverter:** SPH3000
**Pi:** PI Zero 2 W
**PI OS:** Debian Bookworm
**USB to RS485:** USB-RS485-WE-1800-BT (the one you linked to)

666djb, 2024-12-09:
I'm going to have to have a think about this one! It initially looks like RS485 MODBus comms problems such as the inverter not responding. I'm not familiar with mbpoll, but can see what that shows on mine. To check for the obvious, does running solarpi (or mbpoll) as root change anything ("sudo npm start" run from /opt/solarpi)?

tekgrrl, 2024-12-12:
Thanks for giving this some thought. I'll check again soon, but I did have `permission denied` (to the device file) in the logs original, then I added the `solarpi` user to the group that owned the device file and it was OK. I don't think it's that.

Do you know if the setting for `RS485 Addr` under `Basic Parameters` on the Inverter is important? Maybe it should reflect the RS485 port that I'm using?

tekgrrl, 2024-12-21:
Great, thanks for this. I'll reassemble the project and try again. 

BTW: I opened a case with Growatt and they were no help at all (not blaming them, they probably have a very small support team that's overburdened)

nicdd, 2024-12-21:
> Great, thanks for this. I'll reassemble the project and try that.
> 
> BTW: I opened a case with Growatt and they were no help at all (not blaming them, they probably have a very small support team that's overburdened)

Yes, their documentation is a mess. We have to do a lot of "reverse engineering" in order to understand what is what and for which purpose :) it is a nice and expensive hobby. 

My goal would be to automate the behavior of the inverter depending on the battery status, time, weather conditions, so as to reduce the inverter's own consumption as much as possible, putting it on standby or turning it on in the appropriate context. It's a 4 kWp pv-system :).
I also managed to install a Home-Assistant but also to set up raspi to communicate with the inverter, even after a reboot.

I invested I think about 20 days of testing and documentation, I bought cables, 2 raspi and other accessories worth about 200 euros probably so far. I don't even want to calculate how many years it will take for the savings in kwh that I will achieve to amortize these investments - if I thought about that I should have given up a long time ago.

tekgrrl, 2025-01-04:
Working correctly now. I hard wired the RJ45 cable to the USB adapter using pins 4 and 5 on the cable I have, which are the Blue and Blue/White striped wires. I'm using RS485-2 on the Inverter (which is in fact a Growatt SPH-3600) with the RS485 addr (under Basic Parameters) set to COM 001.

I cna now see data from the inverter in Home Assistant, though it's no longer sending data to Growatt. I don't mind that but right now I have no dashboards. I'll investigate and circle back if I can't get it to send data to both.

Thanks for your support @666djb and @nicdd ! 

BTW: The `mbpoll` command that I mentioned still doesn't work which is fine. Just good to know that it's not useful for this use case.

tekgrrl, 2025-01-04:
Seems that once I got SolarPi working, the Growatt Datalogger was removed, presumably it either went offline or some conflict caused it to be removed. I will test that hypothesis, but if I can only have one datalogger, I'll prefer to have SolarPi or Solar Assistant.

666djb, 2025-01-05:
It looks like you've had progress and the root cause was the RS485 connection. I've just been and looked at my inverter and I don't have a menu for changing the RS485 Address. I have coded solarpi so that it takes address "1" (line 33 of solarpi.ts). I can run some tests and see what the inverter uses but suspect it uses address "0" be default. 

It is safe to change the hard coded value (in solarpi.ts) and run npm install to recompile the code. Happy to give more precise instructions to help, and if useful I could make the address a configuration parameter that can be changed more easily.

I'm keen to hear how you get on and to hopefully add the SPH3600 to the known-good list!

(For info, I've previously run my setup to use next day solar forecast values to alter how much my batteries are charged in order to minimise draw from the grid. I now do something different and automatically export any left over power back to the grid late evening whilst charging fully overnight. I also run another automation that detects when I get a car charging half hour slot and set the batteries to charge at those times too in order to take advantage of the cheap rate.

tekgrrl, 2025-01-06:
Thanks for the reply. I'll look at the code and if I parameterize it I'll send you a PR.

One question, as soon as Solar Pi started working (or when I turned it on in Home Assistant), the Growatt software stopped receiving data and could no longer communicate with the inverter. I tried a few things and then disconnected the RS485 cable but it didn't recover. Today I did two things, with the RS485 cable still disconnected, I switched the RS485 setting back to Battery and also restarted the RF dongle. It started working after one if those things, just haven't isolated which.

Am I going to have to choose between SolarPI or Growatt software? The installers need access via Growatt, but I'm probably OK with just giving them access when I need support.

BTW: The inverter was installed with an RF dongle and talks to a 'ShineLab' device that's connected to my internet. If it's the dongle hanging I may try a WiFi dongle.

666djb, 2025-01-09:
I suspect that the reason the Growatt software stopped receiving data might be down to a clash of RS485 addresses causing the dongle to become inactive. Try a different address in solarpi.ts for a test, use anything 2 or 3 etc. I am guessing that the RF dongle talks via RS485 to the inverter (much like SolarPi is doing). The Growatt dongle on my inverter sounds to be different: this connects via WiFi and reports back to China.